Edgenesis/shifu #273: serial device data acquisition

478264002, 2022-09-28:
Can I support serial device data acquisition? For example, weighing machine equipment and light curtain equipment

saiyan86, 2022-09-28:
Hi @478264002, thanks for opening this issue? By serial device, do you mean RS-232/485? If so, you'll need to convert it to Ethernet first. For example, https://www.amazon.com/RS485-High-Speed-High-Stability-Up-gradable-Communicate/dp/B07RQK56TM?th=1

tianranhunda, 2022-09-28:
Thanks for issue. More detais need to be discussed. Could you please connect me in Wechat (if7369),I will invite you to our developer community. We can support the devlepement of Shifu together.

478264002, 2022-09-28:
Yes, there is a solution to my problem, thanks.

closed 2022-09-28 by saiyan86