Edgenesis/shifu #533: Not able to read the OPCUA data, throwing Failed to read message from Server, error: write tcp 10.2xx.0.xx:47902->192.xxx.x.xxx:49320: write: broken pipe

MaheshDevulapally, 2023-03-09:
Hi ,
I'm running shifu locally on my kind cluster. 
When I tried to get the value from my opc uA server ,its throwing above mentioned error ..can anyone help me on this?Attached the below picture 
[image]

tomqin93, 2023-03-09:
Hi @MaheshDevulapally , thanks for this issue
Do you mind sharing your configmap and edgedevice YAML files?
Also it would be great if you can share your OPC UA server's spec

MaheshDevulapally, 2023-03-09:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Hi Tom,

Yeah Sure PFA .

And regarding my OPC UA server I'm using Kepware Server EX which is nothing OPC UA Server running on my local pc itself.

Thank you ,Mahesh.

The information contained in this communication is intended solely for the use of the individual or entity to whom it is addressed and others authorized to receive it. It may contain confidential or legally privileged information. If you are not the intended recipient you are hereby notified that any disclosure, copying, distribution or taking any action in reliance on the contents of this information is strictly prohibited and may be unlawful. If you have received this communication in error, please notify us immediately by responding to this email and then delete it from your system. EY is neither liable for the proper and complete transmission of the information contained in this communication nor for any delay in its receipt.

tomqin93, 2023-03-09:
@MaheshDevulapally Thanks for sharing, we'll give that a try
In the mean time, do you mind sharing your configmap and edgedevice YAML files?
It looks like the pod has crashed due to errors, you can checkout the logs by 
```bash
kubectl logs -n deviceshifu {POD_NAME}
```

MaheshDevulapally, 2023-03-09:
Hi Tom,

PFA files .
Pods are running fine .
***@***.***

Thank you, Mahesh.

MaheshDevulapally, 2023-03-09:
Hey Tom,
Just an update from my side that finally the issue is resolved (wrangled with end point address and deployments)  and receiving my Kepware OPC Ua server as below . But question here is that I have many tags/many devices  on OPC UA server side , so which tag value it is reading?? how can I see/browse my tags?  Also when I use the command get_server it is showing "FreeOpcUa Python Serverroot" but It should be KepwareEXserver right?
***@***.***

Thank You, Mahesh.

tomqin93, 2023-03-10:
Hi @MaheshDevulapally , we are not able to see any of the files/attachment you sent.
The OPC UA deviceShifu maps each HTTP API to a Node ID, for example in https://github.com/Edgenesis/shifu/blob/main/examples/opcuaDeviceShifu/opcua_deploy/opcua_deviceshifu_configmap.yaml 
we have the following:
```yaml
...
  instructions: |
    instructions:
      get_value:
        protocolPropertyList:
          OPCUANodeID: "ns=2;i=2"
      get_time:
        protocolPropertyList:
          OPCUANodeID: "i=2258"
      get_server:
        protocolPropertyList:
          OPCUANodeID: "i=2261"
...
```
so `get_value` maps to `ns=2;i=2` and so on.

Currently we do not support NodeID browsing. May I know what's the use case? You can create a feature request for that.

Regarding the `FreeOpcUa Python Serverroot` from server command, it means you are probably connected to our mock opcua device. You need to configure edgedevice.yaml to your server's address

MaheshDevulapally, 2023-03-10:
Hi Tom,
Sorry I was replying through Email ,that might be the reason you are not able to read the files. Attached now.
Yeah u were right even though I have changed the node address it referred to free opc ua server which i'm running. I deleted the deployment/service now it is redirected to my Kepware Opcua server but below error I'm getting kindly help me 
[image]
Thank you,Mahesh
[Yamlzipfile.zip](https://github.com/Edgenesis/shifu/files/10938221/Yamlzipfile.zip)

MaheshDevulapally, 2023-03-16:
Hi Tom,
Yeah you were right. It was output of default examples.
Now just look at the below output when I'm trying to connect Kepware OPC server which I insatlled locally. Also attached the configmap and edgedevice YAML files.
[image]
[TempMarch16th.zip](https://github.com/Edgenesis/shifu/files/10987202/TempMarch16th.zip). 
Also could you help me how get those node id's any reference article etc...currently I'm using OPC expert tool to know the node IDs... It would be great if shifu could support OPC NodeID browsing.

tomqin93, 2023-03-16:
Hi @MaheshDevulapally 
Thank you for the info, it looks like you are trying to request a node ID that does not exist in your Kepware Server.
You may want to find a valid ID from OPC expert tool and replace it in the default ConfigMap's `OPCUANodeID` field
BTW just saw your issue https://github.com/Edgenesis/shifu/issues/545 and we are looking into it.

MaheshDevulapally, 2023-03-16:
Hi @tomqin93 
Finally I was able to read the data from Kepware server.
But I have below questions 
1. If I want to read multiple tag values ,in which format I should provide OPC UA node IDS? I tried in list format but didn't worked. 
      get_value:
        protocolPropertyList:
          OPCUANodeID: ["ns=2;s=Channel1.Device1.Tag1","ns=2;s=Channel1.Device1.Tag2"]
2. does instructions are interlinked? meaning if I provide wrong identifier for get server or get time it will break get value instruction?
        get_value:
          protocolPropertyList:
           OPCUANodeID: "ns=2;s=Channel1.Device1.Tag1"
      get_time:
        protocolPropertyList:
          OPCUANodeID: "i=84"
      get_server:
        protocolPropertyList:
          OPCUANodeID: "i=85"
3. This is related to MQTT deployment I'm facing below when I'm port forwarding
[image]

MaheshDevulapally, 2023-03-17:
Hi @tomqin93 ,
I have added Getvalue1 instruction by giving second tag Nodeid but its throwing below error .can you help me any config in some other Yaml files missing?
[image]
[image]

3. Same error with 31883

tomqin93, 2023-03-17:
Hi @MaheshDevulapally , when you update the configmap, did you delete the pod?
Currently you need to manually delete the pod or reapply via `kubectl delete -f <folder>` and `kubectl apply -f <folder>` to take effect

MaheshDevulapally, 2023-03-17:
Hi @tomqin93 Yeah this time I forgot to delete pod this time and it worked. Another observation is that even when remote opcua server is down , again we need to delete the pod and create again. 
Regarding MQTT any help ?
also correct me if I'm wrong ,below CURL command we need to run inside nginx right? 
[image]
Also could you tell me right command for windows to below ....grep is not recognized 
sudo kubectl get pods -A | grep robotarm

tomqin93, 2023-03-17:
Hi @MaheshDevulapally ,
1. Regarding the OPC UA server down and not reconnecting, can you please create an issue for this so we can keep track of it? It may be a bug from the library we used https://github.com/gopcua/opcua/issues/434 
2. For MQTT, did you try forwarding without the `--address` prefix?
3. Yes the command looks correct, make sure you have changed the MQTT server address and configmap accordingly as stated here https://shifu.run/docs/guides/cases/connect-mqtt
4. the windows equivalent to should be `sudo kubectl get pods -A | findstr 'robotarm'`. Also if you are running just a few devices, you can just remove the last part and issue `sudo kubectl get pods -A` directly.

tomqin93, 2023-03-17:
Hi @MaheshDevulapally , btw not sure if you saw my comment in the feature request issue.
 To better understand your request and provide you with the best possible solution, could you please send an email to [xqin@edgenesis.com](mailto:xqin@edgenesis.com)?
We can also arrange a short 1-on-1 meeting to discuss the issue in more detail.