Edgenesis/shifu #538: In deviceShifuBase feature, convert it from single EdgeDevice to a list type of EdgeDevice

kris21he, 2023-03-14:
Change EdgeDevice type from single type to array type, only support first item of config so not breaking current behavior.

Accept criteria:

1. in examples/httpDeviceShifu/deployment/http-edgedevice.yaml, let's config as following
```
spec:
  -sku: "device1"
    connection: Ethernet
    address: 192.168.11.16:12345
    protocol: HTTP
```
2. after this change, make shifu still working with following command
```
kubectl apply -f examples/httpDeviceShifu/deployment
```

3. if there are multiple edgedevices in http_edgedevice.yaml file, always use the first one and ignore others

SuperAayush, 2023-04-20:
Hey @kris21he, I would like to work on this!!

tomqin93, 2023-04-21:
Hi @SuperAayush , thank you so much!
I will assign this issue to you. If you ever need anything, feel free to reach out to us, my e-mail address is xqin@edgenesis.com

kris21he, 2023-04-23:
@SuperAayush I am so glad that you picked this up, any questions free free to let us know.
Looking forward to your ideas, thoughts, and PR for this.
Thank you for your enthusiasm of contributing to the shifu community!