Edgenesis/shifu #542: The simplest FSM implementation of deviceShifu

kris21he, 2023-03-15:
Finite State Machine(FSM) could be a useful feature to logically automate deviceshifu's state and behavior.
Here we request to add example of applying FSM to deviceshifu. we can create a folder called "fsm/examples" under /pkg.

To create the simplest FSM as following:
Please use https://github.com/looplab/fsm to define a FSM that can control a LED deviceShifu's ON and OFF state changes.
1. The LED can turn ON if and only if it is in the OFF state
2. The LED can turn OFF if and only if it is in the ON state

bhavyastar, 2023-04-27:
Hey @kris21he , I would like to work on this issue.

tomqin93, 2023-04-28:
@bhavyastar Thank you!
 Feel free to reach out to us on Discord/Slack if you ever need anything

tomqin93, 2023-06-09:
Hi @bhavyastar , any update on this issue? let us know if have any questions.

tomqin93, 2023-06-09:
> Yes, I have 2-3 questions. Are you available for meeting on 11th June?

Sure, you can ping me on Discord to schedule a meeting

tomqin93, 2023-08-28:
unassigning to make this issue available for others