Edgenesis/shifu #545: OPC UA node ID Browing feature on Shifu .Currently we are providing  NodeID' s Manually by going to the OPC UA server loc.

MaheshDevulapally, 2023-03-16:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

**Additional context**

Add any other context or screenshots about the feature request here.

MaheshDevulapally, 2023-03-16:
I have faced this issue while I'm fetching data from kepware OPC UA server. that server is having 100's of devices which is hard to get all those node Id's Manually

tomqin93, 2023-03-16:
Hi @MaheshDevulapally, thank you for creating this feature request!
Can you please provide more detail on the request by filling in the template above?

tomqin93, 2023-03-16:
Hi @MaheshDevulapally , thank you for updating the issue. We have added it to our queue with high priority to ensure a timely resolution. 
To better understand your request and provide you with the best possible solution, could you please send an email to [xqin@edgenesis.com](mailto:xqin@edgenesis.com)? 
We can also arrange a short meeting to discuss the issue in more detail.

cburkovich2015-collab, 2026-11-06:
OPC UA integration becomes especially challenging in industrial environments with hundreds of devices and thousands of data points. Manually configuring Node IDs can quickly become difficult to manage, especially when new equipment is added or configurations change.

A scalable industrial IoT solution should include automated device discovery, centralized tag management, protocol integration, and reliable edge data collection. Connecting this data with existing MES, ERP, and SCADA systems is equally important to ensure operational insights can be used beyond basic monitoring.

When standard gateways or platforms cannot meet specific requirements, custom industrial IoT development can help extend their capabilities through tailored integrations: https://softdoes.com/industries/construction/industrial-iot-software-development