Edgenesis/shifu #547: The traffic light FSM implementation for deviceShifu

kris21he, 2023-03-17:
**Is your feature request related to a problem? Please describe.**

I want to use FSM to represent the traffic light change logic, take the deviceShifu as the datatwin of the traffic light

**Describe the solution you'd like**

1. The traffic light has RED, YELLOW, GREEN states
2. The light can change from RED to GREEN
3. The light can change from GREEN to YELLOW
4. The light can change from YELLOW to RED
5. The light can not change any other state except above rules

**Additional context**
use https://github.com/looplab/fsm to implementation
refer to https://github.com/Edgenesis/shifu/issues/542 if that was implemented

huerni, 2023-05-27:
May I attempt to address this issue?

tomqin93, 2023-05-27:
Thank you @huerni , absolutely! 
I will go ahead and assign this issue to you. 
Let us know if you need anything, you can reach me via email at xqin@edgenesis.com

ashutosh887, 2023-06-16:
Is this issue resolved @tomqin93 @huerni ?
I would like to work on this

huerni, 2023-06-17:
> Is this issue resolved @tomqin93 @huerni ? I would like to work on this

Resolving