Edgenesis/shifu #597: path error while setting up shifu demos in local system

SuperAayush, 2023-05-10:
**Describe the bug**

While trying the shifu demos from here: https://shifu.dev/docs/tutorials/demo-try

I see that the AGV device works fine after setting up shifu in the local system following this guide: https://shifu.dev/docs/tutorials/demo-install

According to my knowledge, AGV works fines because it comes preinstalled with shifu but other devices need to hit the correct file to work fine.

All the other devices apart from AGV require users to get into the demo directory named **testdir** installed after running:
`curl -sfL https://raw.githubusercontent.com/Edgenesis/shifu/main/test/scripts/shifu-demo-install.sh | sudo sh -`

**To Reproduce:**

Steps to reproduce the behavior:

1. Setup shifu using - [Quick Install](https://shifu.dev/docs/tutorials/demo-install)
2. Try to run any device apart from AGV - [Try out](https://shifu.dev/docs/tutorials/demo-try)

**Solution:**

Adding a command to change the directory to `testdir`after completing the quick install guide.

Starting lines of the preparation section can have:

To try out the devices mentioned below make sure to install shifu in your local system from [Quick Install](https://shifu.dev/docs/tutorials/demo-install) before proceeding forward.

After the completion of Quick Install guide make suer to change the directory to `testdir` using:

`cd testdir`

Section: 

[image]

**Server:**

 - Shifu Version: [e.g. v0.0.5]
 - Kubernetes distro - KinD
 - CPU Architecture - arm64

**Additional Views**

The directory name should be changed from testdir to something like shifudemos, having usual names like testdir might trouble in local implementation and also navigating the directory easily.

tomqin93, 2023-05-12:
@SuperAayush  Thank you for this issue and solution! We'll be making this change in our docs repo https://github.com/Edgenesis/shifu.run

SuperAayush, 2023-05-12:
Hey @leepala!!

Let me know if any help is needed from my end if you are comfortable I can take up this issue and go for a PR for you to review 😄.

Thank You!!

leepala, 2023-05-12:
Hey @SuperAayush, thanks for your question. We welcome your help in solving this problem! 
All of Shifu's documents are in this repo https://github.com/Edgenesis/shifu.dev, welcome to make any changes and make PR.
The directory name does need to be updated, thanks for the suggestion, I think `shifudemos` is all right . If you want to fix it, you can update [this shell script](https://github.com/Edgenesis/shifu/blob/main/test/scripts/shifu-demo-install.sh) to do it.

SuperAayush, 2023-05-12:
> Hey @SuperAayush, thanks for your question. We welcome your help in solving this problem! All of Shifu's documents are in this repo https://github.com/Edgenesis/shifu.dev, welcome to make any [...]

The current name of the directory is **testdir**, should we change it to **shifudemos** ?
testdir is very usual name will be very hard to navigate.

leepala, 2023-05-12:
yep, let's change it to **shifudemos** instead of **testdir**☺️

SuperAayush, 2023-05-18:
> Hey @SuperAayush, thanks for your question. We welcome your help in solving this problem! All of Shifu's documents are in this repo https://github.com/Edgenesis/shifu.dev, welcome to make any [...]

To make the change in the directory name from testdir to shifudemos, should I create a separate issue because:
- changes have to be made in this repo itself
- it is a good open-source practice to have a separate issue for uncommon changes so that it is easier to refer in future requests.

tomqin93, 2023-05-22:
> estdir to shifudemos, should I create a separate issue because:
> 
> * changes have to be made in this repo itself
> * it is a good open-source practice to have a separate issue for uncommon changes so that it is easier to refer in future requests.

Hi @SuperAayush, sure please go ahead and create a separate issue for the renaming.

closed 2023-05-30 by tomqin93