Edgenesis/shifu #790: Shifu Controller should automatically create telemetryService when TelemetryService CRD presents

tomqin93, 2023-12-21:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
Currently TelemetryService requires a Deployment and Service with [deployment YAML](https://github.com/Edgenesis/shifu/blob/main/pkg/telemetryservice/install/telemetryservice_install.yaml ) manually.

As a Shifu developer, I'd like this to happen automatically. When telemetryService CRD has been detected, Shifu Controller should automatically deploy the deployment manifests and vise versa

**Describe alternatives you've considered**

Doing it manually

**Additional context**

lizzy-0323, 2024-10-21:
Hi, @tomqin93, I would like to work on this issue

tomqin93, 2024-10-21:
@lizzy-0323 Thank you! You can assign this issue to yourself.
Also please contact info@edgenesis.com for your freebies!

closed 2024-12-19 by rhoninl