Edgenesis/shifu #1115: Shifu should check if <updateIntervalSec> in Gateway is a positive integer.

Twilikiss, 2025-01-24:
In the deployment deviceShifu cooperate LwM2M Gateway when used in the corresponding Edgedevice configuration `GatewaySettings. LwM2MSetting.UpdateIntervalSec` for a non positive integer (0 / - 1/2/3, etc.), Still able to deploy and run successfully.

[image]

[image]

Chi-Kai, 2025-02-06:
I want to try this issue.

rhoninl, 2025-02-07:
@Chi-Kai  Thank you for your interest in contributing! I’ve assigned the issue to you and am confident you’ll do a great job. If you have any questions or need further clarification along the way, please don’t hesitate to reach out.

PR #1144 (2025-03-12): [issue #1115] check if <updateIntervalSec> in Gateway is a positive i…
…nteger.

**What this PR does / why we need it**:
fix #1115 
**Will this PR make the community happier**? 

**Which issue this PR fixes** *(optional, in `fixes #<issue number>(, fixes #<issue_number>, ...)` format, will close that issue when PR gets merged)*: fixes #

**How is this PR tested**
- [ ] unit test
- [ ] e2e test
- [x] other (please specify)
[image]

**Special notes for your reviewer**:

**Release note**:

```release-note
```